Ultimaker/Cura #17459: Reflect Layer Color Change on Thumbnails

JaneX8, 2023-11-28:
### Is your feature request related to a problem?

As described here: https://3dprinting.stackexchange.com/questions/21587/is-there-a-way-to-reflect-filament-color-change-in-cura-generated-thumbnails and here https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/issues/64. Cura created Thumbnails are always one-color only. It does not reflect the 'Color Change at Layer' script in the generated thumbnails.

### Describe the solution you'd like

I want to create multi-color thumbnails that keep into account the 'Filament Change' script that was used, for example, when printing something as shown below: 

[image]

### Describe alternatives you've considered

Generating thumbnails outside of Cura using other methods and embedding it inside the G-code.

### Affected users and/or printers

Anyone using Thumbnail generation in combination with Color Layer Change.

GregValiant, 2023-11-28:
I'm not aware of "Color Change at Layer"  There is "Filament Change" and "Pause at Height" that allow for color changes.

Both of those are "post-processors" so the work they do is at "save time" and after slicing has completed.  That means that when the model is sliced and previewed your color changes don't exist.  Consequently, the snapshot that "Create Thumbnail" takes of the Cura build plate doesn't include the color changes (because they aren't there).

I'm not saying it's impossible to do this, but I think it's close.

JaneX8, 2023-11-28:
I meant 'Filament Change' as shown on the screenshot in the linked post.

It would also require the plugin to specify to what color the filament will be changed, this can also be used for showing a printer message by the way, 'change to blue', 'change to red'.

What do you think is the right approach to achieve this?