Ultimaker/Cura #17460: 368 copies of model cannot be sliced; also GUI slows down

dimitsev, 2023-11-28:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Ultimaker 7

### Describe model location

My model is small and slices without problem. The slicing fails when I multiply my model to fill the entire space with 368 identical copies. Also, the Ultimaker GUI becomes really slow when I have that many copies.

### Describe your model

My model is small and slices without problem. The slicing fails when I multiply my model to fill the entire space with 368 identical copies. Also, the Ultimaker GUI becomes really slow when I have that many copies.

### Add your .zip here ⬇️

[UMS7_fiber_clamp_x368.zip](https://github.com/Ultimaker/Cura/files/13486396/UMS7_fiber_clamp_x368.zip)

GregValiant, 2023-11-28:
Thanks for the report.
In regards to the slow operation - you've given your computer a lot to think about.
I was able to slice this in 5.2.2 but not with later versions.

I'll mark this as a duplicate slicing error and the Cura team will take a look.
My Cura log from 5.6beta shows:
2023-11-28 07:21:45,620 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-11-28 07:21:45.626] [error] Plugin 'CuraEngineGradualFlow' running at [ipv4:127.0.0.1:56569] for slot GcodePathsModifyService failed with error: Received message larger than max (4631869 vs. 4194304)
2023-11-28 07:21:46,676 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1027]: Backend exited abnormally with return code 3221226505!
2023-11-28 07:21:46,697 - DEBUG - [MainThread] cura.BackendPlugin.stop [130]: EnginePlugin: CuraEngineGradualFlow was killed. Received return code 1
2023-11-28 07:21:46,699 - DEBUG - [MainThread] cura.BackendPlugin.stop [130]: EnginePlugin: CuraEngineTiledInfill was killed. Received return code 1
2023-11-28 07:21:46,700 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [410]: Attempting to kill the engine process
2023-11-28 07:21:46,701 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [416]: Killing engine process
2023-11-28 07:21:46,701 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [419]: Engine process is killed. Received return code 3221226505
2023-11-28 07:21:46,702 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [294]: Previous socket existed. Closing that first.

This is in 5.2.2.  I set the brim width to "3" so the entire build plate would print (else the outside most parts would be excluded).
[image]

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant