Ultimaker/Cura #17461: sliciing fail on Ender3v3 SE

kyawnaingtun89, 2023-11-28:
### Cura Version

5.6.0 beta 2

### Operating System

windows 11

### Printer

Creality Ender3V3 SE

### Name abnormal settings

Tree support cause the error.

### Describe model location

middle and front

### Describe your model

it is name plate with footing. I need tree support

### Add your .zip here ⬇️

[prototype2 - name plate.zip](https://github.com/Ultimaker/Cura/files/13487564/prototype2.-.name.plate.zip)

GregValiant, 2023-11-28:
Thanks for the report.
I used 5.6.0 stable and it sliced.  You might try rotating the model or changing the Initial Layer Height by a little bit.
That should print without any supports.  Your printer should bridge that little gap with no problem.  You could enable the Bridge Settings.  It doesn't appear to really need that though.  I'd print two at once.  The printer will be hitting the minimum layer height and slowing down so two won't hurt.  You could turn off supports for one of them just to check.

I'm going to leave this open and I'll mark it as a duplicate slicing error.  The Cura team has lots of moons and crystal dragons but a small problem child like this might have some value in debugging (if they can duplicate the problem).

[image]

5axes, 2023-12-07:
No issue, and I don't think we need some support. But could be an issue with Windows 11.. We will never know.
[image]

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant