Ultimaker/Cura #17463: The slicing; it failed.

GillaMonstah, 2023-11-28:
### Cura Version

5.4.0

### Operating System

Windows 10, Intel i7-7700K 4.2GHz

### Printer

Anycubic i3 Mega (S, Pro)

### Name abnormal settings

Print speed +10 and travel speed +20, supports enabled (63 deg. overhang), build plate adhesion set to skirt.  I gets further on default, but still fails

### Describe model location

If I rotate it 90 deg. x-axis, the slicing stalls

### Describe your model

This tool does not appear to exist anymore

### Add your .zip here ⬇️

https://drive.google.com/file/d/107n9cTVH5E_P9pmZFAX-FFBrGNTTsRgs/view?usp=sharing

GregValiant, 2023-11-28:
Thanks for the report.
It wouldn't slice for me in 5.5.0.  I changed the Initial Layer Height to 0.25 and it sliced.  You might also try turning of "Print Thin Walls" which seems to be related to the slicing bug.
I'll mark this as a duplicate report and go ahead and close this.  Moon models have been here numerous times and the Cura team is aware of this problem.  I hope you understand.
[image]

closed 2023-11-28 by GregValiant

GillaMonstah, 2023-11-30:
I changed the initial layer height to 0.25 and it sliced for me as well.  Thanks for your help!