Ultimaker/Cura #17465: Print stair steps and then starts all over again in the correct position

floridajohn, 2023-11-28:
### Cura Version

5.5.0

### Operating System

Windows !0 Professional

### Printer

Creality Ender 5 Plus with TH3D screen and UFW 2.74a

### Reproduction steps

print walks over consistently in the Y axis but then at about 2mm high returns to the correct postion
	starts the "stair step" all over again
	X-axis is fine

stl file
	created with Tinkercad - it's just some numbers with a ledger below them
	I've printed many of these in the last year and never had a problem
	printed this same .stl  months ago and worked fine (file is gone)
	previous tries - resize smaller original in CURA to 250% x 250% x 150%
	problems in the last week with elements both ungrouped and grouped 
	this try - increase file in Tinkercad to finished size, group all and rotated to final orientation
		only had to nudge it over to the left to get it out of the center
	use Elmer's washable school glue as an interface between the glass plate and the print
	I can also successfully print 150mm x 300m items with or without glue

### Actual results

same problem
I've never had anything like this before

### Expected results

not stair step while printing

### Add your .zip and screenshots here ⬇️

[5-5-0_cura_settings.zip](https://github.com/Ultimaker/Cura/files/13491971/5-5-0_cura_settings.zip)
[BRN on plate with problem.zip](https://github.com/Ultimaker/Cura/files/13491972/BRN.on.plate.with.problem.zip)
[FS BRN 06561 SQ E5 MS BRN 230 80 50 cubic 000000 100 100 100.zip](https://github.com/Ultimaker/Cura/files/13491973/FS.BRN.06561.SQ.E5.MS.BRN.230.80.50.cubic.000000.100.100.100.zip)
[FS BRN 06561.zip](https://github.com/Ultimaker/Cura/files/13491974/FS.BRN.06561.zip)

GregValiant, 2023-11-28:
Thanks for the report.
I've never seen this problem and I can't reproduce it.
You can open that Gcode file in Cura and see if the steps exist in the Gcode.  My guess would be that they aren't there.  This is the kind of problem that if it was truly Cura's fault then nobody would use it.
If you want you can post the gcode here and I'll take a look.
The gcode reader in Cura is a separate app from the slicing engine so you get a true view of the gcode.

floridajohn, 2023-11-29:
I ran the file using CURA 5.5.0 standard quality on both the E5+ and my
older Ender 3.  The 3 was fine which leads me to conclude it's not a CURA
problem.  Close it?

It's such a perfect step that I thought it had to be CURA.  Need to figure
out why the Z-axis steps so nicely forward..

closed 2023-12-07 by GregValiant