Ultimaker/Cura #17467: Slicing failed after a few seconds.

rbenites, 2023-11-28:
### Cura Version

5.5.0

### Operating System

Windows 11 Home

### Printer

Creality Ender 3 S1 Pro

### Name abnormal settings

using CURA preset config

### Describe model location

Slicing worked AFTER using the auto orientation extension 2 times.

### Describe your model

I have sliced this before with no issues and was able to slice after. Not sure why it failed this time.

### Add your .zip here ⬇️

Zip file is too large (over by 1 mb).
i added the zip to my google drive if anyone needs access - https://drive.google.com/drive/folders/1y78X3s7EmLdq6OQw9oQnCWd-hbOt3DVa?usp=sharing

5axes, 2023-12-05:
Duplicate, Duplicate , Duplicate, Duplicate and Duplicate ... **Print Thin Walls**
[image]

MariMakes, 2024-01-18:
Hi @rbenites,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 5 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17679

closed 2024-01-18 by MariMakes