Ultimaker/Cura #17468: TL-D3 Pro

djwest9999, 2023-11-28:
### Is your feature request related to a problem?

i wish i could use the new version for my TL-D3 pro from tenlog but  yall took it off after the update so i have to use the older version and it is super sluggish i have several printers and they all work on the new update but the tenlog wont. and i shouldn't have to download a custom setting it should be there.

### Describe the solution you'd like

add the tenlog back to the new settings. it is not an old machine, and it has a lot of followers.

### Describe alternatives you've considered

trying to add custom setting is very tedious as well you can't get it 100% so i have to use the older version which runs super slow.

### Affected users and/or printers

the tenlog group that is on Facebook as well as all the users that won't complain.

Asterchades, 2023-11-30:
As near as I can figure, this is not the result of the definition being removed - it was simply never incorporated into the master Cura builds. The closest it ever came to happening was a pull request from 2021 (https://github.com/Ultimaker/Cura/pull/9910) which was ultimately closed before being merged due to issues with the definitions not being addressed. No builds going back as far as 4.9 (the most recent build before this request started) seem to include these files in any state.

_Note that a successful pull request is required for any third party printer to be added to the master Cura builds. The Cura team simply doesn't have the time nor the appropriate hardware to create profiles for non-Ultimaker printers, and relies exclusively on community (including vendors themselves) submissions to add these._

Between this and what I've been able to find out elsewhere I suspect this "older version" (most probably 4.8) was installed from the provided SD card. As such this build was likely customised by Tenlog themselves to include the definitions.

Tenlog does have [some instructions](https://www.tenlog3d.com/blog/how-to-set-up-the-latest-cura-for-tenlog-dmp-3d-printer_b21) on how to set up the printer within Cura, along with a Curaprofile (print settings) to be used afterwards. As near as I can figure these instructions - and the profiles - should still be compatible with the latest Cura (5.6), and should likely work fine built on top of the Custom FFF Printer definition.

It would also be technically possible to import the older definitions into a newer Cura build from the customised version. However there's no guarantee that they would work due to the changes that have occurred over the years (particularly with the introduction of Cura 5.0). You would need to find the appropriate JSON file/s in the Resources directory and copy it to the newer installation (I would recommend copying it to the Resources directory found in the Configuration Folder, rather than the parent Cura directory).

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant