Ultimaker/Cura #17470: problema stl fusca

lrivas98, 2023-11-29:
### Cura Version

5.5

### Operating System

windows 10 

### Printer

ender 3 pro

### Describe your model

primera vez usando el archivo

### Add your .zip here ⬇️

[FabConvert.com_mate+fusca+1.zip](https://github.com/Ultimaker/Cura/files/13495102/FabConvert.com_mate%2Bfusca%2B1.zip)

5axes, 2023-12-06:
Model Issue ... 
[image]

But no issue to slice this model ...

[image]

We can close this issue

lrivas98, 2023-12-06:
Utilizo fusion para cortar el modelo? elimina los elementos flotantes y sigo sin poder segmentarlo

5axes, 2023-12-06:
Without Project file , impossible to analyse

lrivas98, 2023-12-06:
[https://www.tinkercad.com/things/aiHV9Sq0nEw-fabulous-snaget-elzing/edit?sharecode=kVIYyGxYMTvH8vFTs-ThI6lEYh7s_v9VVBsL01H7Q7Q ](url)

fieldOfView, 2023-12-06:
Those links to tinkercad are not project files. A project file can be created by loading your model into Cura and using `File -> Save project...`

lrivas98, 2023-12-06:
[image]

lrivas98, 2023-12-06:
[CE3S1PRO_mate fusca 1.zip](https://github.com/Ultimaker/Cura/files/13583362/CE3S1PRO_mate.fusca.1.zip)

lrivas98, 2023-12-06:
[CE3S1PRO_mate fusca 1.zip](https://github.com/Ultimaker/Cura/files/13587158/CE3S1PRO_mate.fusca.1.zip)

5axes, 2023-12-06:
On more Duplicate for **Print Thin Walls** And you will never success to print such part without support.
[image]

lrivas98, 2023-12-06:
Nunca lo pude imprimir porque no me dejaba segmentarlo aunque le pusiera soporte , en resumen debo desactivar IMPRIMIR CON PAREDES DELGADAS y agregar soporte solamente?

5axes, 2023-12-06:
Issue could be generate by this area :

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant