Ultimaker/Cura #17473: Leg Brace Assembly Print MK II

HerculesGrytpypeThynne, 2023-11-29:
### Cura Version

5.5.0

### Operating System

Windows 10 Pro

### Printer

Ultimaker-S7-1

### Name abnormal settings

 Can fail to slice either PLA or PET CF, no known odd settings

### Describe model location

No

### Describe your model

Kit of parts, assessing PET CF vs Tough PLA

### Add your .zip here ⬇️

[UMS7_M35+ Vertically Clamped Leg Brace Whole Assembly Print MKII_110923.zip](https://github.com/Ultimaker/Cura/files/13499403/UMS7_M35%2B.Vertically.Clamped.Leg.Brace.Whole.Assembly.Print.MKII_110923.zip)

5axes, 2023-12-07:
Position or orientation can change the success of this print.

[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0