Ultimaker/Cura #17477: CE3V3SE_Short +2 deg_1

agriffinengr, 2023-11-29:
### Cura Version

5.6.0

### Operating System

Windows 11 Pro

### Printer

Creality Ender-3 V3 SE

### Name abnormal settings

Slicing failure

### Describe model location

Center. Changing rotation nor position mattered.

### Describe your model

Rotor blade for hydrokinetic turbine test

### Add your .zip here ⬇️

[CE3V3SE_Short +2 deg_1.zip](https://github.com/Ultimaker/Cura/files/13502879/CE3V3SE_Short.%2B2.deg_1.zip)

agriffinengr, 2023-11-29:
Also failed on 5.5.0 and 5.4.0 but i previously successfully sliced this on one of these. I was re-slicing to put on a different flash drive.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic