Ultimaker/Cura #17480: Failed Slicing Error

eschwitter, 2023-11-30:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Creality Ender 3 Pro

### Name abnormal settings

It slices at default if I move the model

### Describe model location

Yes it slices if I move/rotate the model

### Describe your model

No, this is my first time slicing it.

### Add your .zip here ⬇️

[dicemug_v1.10_snug_fit.3mf.zip](https://github.com/Ultimaker/Cura/files/13508048/dicemug_v1.10_snug_fit.3mf.zip)

5axes, 2023-12-07:
transfer expired

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic