Ultimaker/Cura #17484: Slicing on MX Linux failed after upgrading 5.5 and 5.6.0: Remote exception on Slot 'InfillGenerateService'

Neelixx, 2023-11-30:
### Cura Version

5.6.0

### Operating System

MX Linux

### Printer

Voron 2.4 350

### Name abnormal settings

Ironing?

### Describe model location

Model was automatically arranged 

### Describe your model

STL is from Voron repository, I tested various models which worked before. Slicing is also possible on Windows 10.

### Add your .zip here ⬇️

[V350_[a]_keystone_blank_insert.3mf.zip](https://github.com/Ultimaker/Cura/files/13516296/V350_.a._keystone_blank_insert.3mf.zip)

[stderr.log](https://github.com/Ultimaker/Cura/files/13516327/stderr.log)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/13516332/cura.log.zip)

Till version 5.4 I had no problems. I also did a fresh install, removed the configuration folder, some some parameters for Voron 2, then the crash occurred.
In the log files I found:
DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-11-30 19:07:39.667] [error] failed to connect to all addresses; last error: UNKNOWN: ipv4:127.0.0.1:45481: Failed to connect to remote host: Connection refused
2023-11-30 19:07:39,669 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [140]: [Backend] what():  Remote exception on Slot 'InfillGenerateService': failed to connect to all addresses; last error: UNKNOWN: ipv4:127.0.0.1:45481: Failed to connect to remote host: Connection refused
2023-11-30 19:07:39,670 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1027]: Backend exited abnormally with return code -6!

Neelixx, 2023-11-30:
It seems when I uninstall the new component "CuraEngine Tiled Infill Generation" then slicing is possible again.

jellespijker, 2023-11-30:
Duplicate of https://github.com/Ultimaker/CuraEngine_plugin_infill_generate/issues/7

Unfortunately the plugin isn't currently support on your system. We have a ticket for this on the backlog CURA-11415 for now the only way to get it to work would be to upgrade your system

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic