Ultimaker/Cura #17486: Slicing fails when supports are on nondeterministicly

dobbleg1000, 2023-11-30:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3 S1 plus

### Name abnormal settings

Tree Supports on

### Describe model location

no

### Describe your model

It slices without supports its watertight

### Add your .zip here ⬇️

adds support material in the middle when it does work
[image]
[CE3S1_Malt_rockbearer v2.zip](https://github.com/Ultimaker/Cura/files/13518809/CE3S1_Malt_rockbearer.v2.zip)

GregValiant, 2023-12-01:
Thanks for the report and the file.
I can duplicate this with your project file and with that model in a fresh project.  I didn't find any errors in the model.
With "Tree" supports enabled and the Support Infill Density > 0 the problem occurs.  I was getting hints that the "Dynamic" quality profile might be a problem but it wasn't consistent.  Switching to one of my own profiles sometimes produced the problem as well.
If I change the Initial Layer Height or Layer Height the problem layer# moves up or down but the effect seems to stay at the same Z.
Here I've played with it a bit but the effect is still there.  It looks like "Support Horizontal Expansion" might be involved as well.  
This is the first report I've seen of this behavior.
@MariMakes and the Cura team will take a look.
[image]

dobbleg1000, 2023-12-02:
Worth noting the slice failing outside of this issue was caused by a bios issue with my 13900k on the MAG Z790 TOMAHAWK WIFI
Updating to latest bios fixed it and i sliced like 3 times in a row.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic