Ultimaker/Cura #17492: 2nd file slicing fail

Jdjfha, 2023-12-02:
### Cura Version

5.60

### Operating System

windows 11

### Printer

anycubic kobra max

### Name abnormal settings

no, I only have this problem with complex files. version 5.5 does has the same problem. I can slice the same files without any problems in version 5.4

### Describe model location

It does not slice in any orientation or location

### Describe your model

yes, with version 5.4, yes, yes

### Add your .zip here ⬇️

[Mini light Globe cap - 4655800.zip](https://github.com/Ultimaker/Cura/files/13532768/Mini.light.Globe.cap.-.4655800.zip)

5axes, 2023-12-06:
Strange basic Shape ... No issue
[image]

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant