Ultimaker/Cura #17494: Slicing Failed after Clean Install upgrading from Cura 5.5.0 to 5.6.0

spiegeleiBr0t, 2023-12-02:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

EigenBau3dSpritzer

### Name abnormal settings

Did not Change any Settings

### Describe model location

Always fails slicing, nothing changes, even if i remove 3 of 4 object and rotate it in any direction

### Describe your model

I did sliced it succesfully in cura 5.5

### Add your .zip here ⬇️

[CFFFP_tail.zip](https://github.com/Ultimaker/Cura/files/13536171/CFFFP_tail.zip)

5axes, 2023-12-18:
An other part position issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic