Ultimaker/Cura #17495: Reloading multi-body 3mf file incorrectly replaces all bodies with copies of a single one

rutgersemp, 2023-12-02:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender S1 Pro running klipper

### Reproduction steps

1. Export mesh from Fusion 360 (3mf, single file structure, high refinement)
2. Load into Cura
3. All separate components are correctly loaded
4. Export and overwrite file again from Fusion 360 (no actual changes necessary)
5. Cura prompts for file-reload, choose to reload file

### Actual results

All bodies get replaced by an equal amount of copies of a single one of body

### Expected results

All bodies are individually updated to reflect changes in the file

### Add your .zip and screenshots here ⬇️

[image]
[image]
[test.3mf..zip](https://github.com/Ultimaker/Cura/files/13536419/test.3mf.zip)

GregValiant, 2023-12-07:
Thanks for the report.
I can't duplicate this.  With two models on the build plate, if I edit one and then re-export it from AutoCad Cura then notices and I get the message.  When I reload the file everything is correct with my updated model and the other models remain unchanged.

I've been playing with One at a Time mode.  I've noticed that regardless of the filenames of the models on the build plate when I save, close and reopen the project file - all the models now have the name of the project file.  So the models are part of the 3mf as single models (within 3Dmodel.model) all with the same name.  

Those models are small.  Your project file doesn't indicate that you were using One at a Time, but did you happen to have it enabled when the problem occurred?  I ask because the object list has them as "test.3mf" and "test.3mf(1) which would seem to indicate One-at-a-Time mode because the models are certainly different but they have the same name.  If they have the same name (test.3mf) then I would expect loading a single will would affect them all.

rutgersemp, 2023-12-08:
Hi Greg,

Just as a sanity check I duplicated it for myself again (have since reboot computer etc.) and still able to do so on my end. I'm also able to duplicate it in older versions (oldest I've tried is 5.2.1)
An interesting aside - I've discovered that if not reloading models through the file change prompt, but doing it manually (right click, reload all models) it behaves as expected and does not display the bug.

I'm printing all at once, I've included a project file which I believe includes all of my slicing settings?
[CE3PRO_test.zip](https://github.com/Ultimaker/Cura/files/13613064/CE3PRO_test.zip)
ce.

I've also checked if there's a difference between exporting multiple bodies or multiple components from Fusion 360, but the behavior is the same.

GregValiant, 2023-12-09:
3mf files have location data in them where STL files do not.  When Cura loads a 3mf it respects that location data.  If the model was at the origin when it was exported that's where it will show up.  If the 3mf happened to be a bumper hitch for a car and was exported from body position then it would end up WAAY over on the right someplace.  The same piece as and STL would show up in the middle of the build plate.
The Cura 3mf project files are like zip folders and you can open one with an app like 7zip to view the individual files.

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant