Ultimaker/Cura #17498: [5.6] Random travel moves while printing

Glav0nja, 2023-12-02:
### Cura Version

5.6

### Operating System

Windows 10

### Printer

Alfawise U30

### Reproduction steps

I am trying to print a 4 in a line board game. To print the base I am using a 0.6mm nozzle. I tried using default settings but I the issue happened again.

### Actual results

While printing the printer would do travel moves instead of print moves. Meaning a fast move without any extrusion. Here is a video showing the random travel moves ruining my print: https://we.tl/t-fjEoUXJ0N1

### Expected results

I want to print without sudden travel moves instead of normal print moves.

### Add your .zip and screenshots here ⬇️

[AU30_base.zip](https://github.com/Ultimaker/Cura/files/13536980/AU30_base.zip)

GregValiant, 2023-12-04:
This is probably a "USB Printing" problem.  It is not maintained as no UltiMaker printers use it and the amount of different hardware combinations it has to talk to make it unlikely that it will ever work for everyone.  It is included with Cura as UM 2 printers require it so it is legacy software.  Essentially, if it works, it works and if it doesn't, it doesn't.
There was an effort to make some changes.  If you go to the Cura MarketPlace and click on the "gear" icon you can scroll down the list and disable USB Printing.  Then click on the Plugins link and scroll down and enable "USB Printing NG".  Maybe using the re-worked version will make a difference.
If the communication isn't smooth between the printer and the print-server (Cura in this case) then you can get buffer over-runs in the printer and that means lost commands.  The other problem is buffer under-runs and you get herky-jerky movement and blobbing.  Pronterface/PrintRun has the same problem.  OctoPrint seems to work quite well.

Glav0nja, 2023-12-07:
SOLVED!
Finally got the printer to print from an SD card (it's a 64gb microSD so i couldn't just format it to fat32 with windows x) had to use a special tool for that).
Printed the whole thing for 4 hours and it worked perfectly thank you so much for the help!
Also thank you for the extensive explanation about the printer-PC communication. You really know your stuff!
I'll also try again with USB and the plugin that you mentioned and OcroPrint just for the sake of it. But from now I'll be printing from SD card anyways :D

closed 2023-12-07 by Glav0nja

GregValiant, 2023-12-07:
Octoprint seems to work well.  For me, the SD card is easy to understand and it works.  My printer sits right in front of my workstation so it isn't like I'm putting a lot of miles on my Chuck Taylors running back and forth from someplace.