Ultimaker/Cura #17499: Slicing failed with unexpected error with known good stl

silverdazed, 2023-12-03:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3 v2

### Name abnormal settings

N/A

### Describe model location

Model does not slice even when changing orientation of model

### Describe your model

No, Model downloaded from online, cannot fix with meshtools, Model is water tight

### Add your .zip here ⬇️

[ICHIGO CASCO F.zip](https://github.com/Ultimaker/Cura/files/13537822/ICHIGO.CASCO.F.zip)

GregValiant, 2023-12-03:
Thanks for the report.
I downloaded the zip folder but it's empty.

silverdazed, 2023-12-03:
That's odd, I'll send it as an email attachment later tonight when I get
the chance if that's alright.

GregValiant, 2023-12-04:
Email attachments won't post here (it's a Github thing) so post the file directly here by dragging and dropping or clicking and adding.

silverdazed, 2023-12-04:
[ICHIGO CASCO F.zip](https://github.com/Ultimaker/Cura/files/13550365/ICHIGO.CASCO.F.zip)
I've included the original STL as well as a screenshot of what settings are included in the project file in case for some reason this time it doesn't attach the actual project file in the archive.

5axes, 2023-12-05:
Duplicate , Duplicate , Duplicate , Duplicate ....   **Print Thin Walls**
[image]

silverdazed, 2023-12-05:
Sorry I'm not quite sure what you mean here by duplicate. And did I have
print thin walls off or do I need to have it on?

5axes, 2023-12-05:
Duplicate means that the issue is often reported and that it comes from the **Print Thin Walls** parameter, which must be inactive to avoid this problem.

silverdazed, 2023-12-05:
Ahhh okay. Makes sense, I'll try it with that off then

MariMakes, 2024-01-18:
Hi @silverdazed, 

Quick update from our side 👋 
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪 

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 8 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others. 
Hopefully we can find a solution soon 🤞 

I'll be closing this issue as a duplicate but you can follow the progress here https://github.com/Ultimaker/Cura/issues/16417

closed 2024-01-18 by MariMakes

silverdazed, 2024-01-18:
Awesome, thank you. Glad to help with data for improvements :)

HellAholic, 2024-02-16:
Small update,
Issue is resolved in CURA-11570 and will be part of the next Cura release.