Ultimaker/Cura #17501: Bad gcode produced when interpreting {code1, code2} in M140, M109, M190

bellzw, 2023-12-03:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

BIBO 2

### Reproduction steps

I have created my own definition for this printer and it has been working for me since before 5.0.  Below is the startup code copied from the Machine Settings; it correctly reflects what's in the json file:
; Start of BIBO starutp code
M140 S{material_bed_temperature, initial_extruder_nr}
M104 T0 S{material_standby_temperature, 0}
M104 T1 S{material_standby_temperature, 1}
G21  ;metric values
G90  ;absolute positioning
M107  ;start with the fan off
T0
G92 E0
G28
G1 Y0 F1200 E0
T1
G92 E0 ;set extruder 2 position to 0
M109 T0 S{material_standby_temperature, 0}   ; now wait for E1  to reach temperature
M109 T1 S{material_standby_temperature,1}   ; now wait for E2  to reach temperature
M109 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr}  ; now set the correct extruder to the correct initial temperature
T{initial_extruder_nr}
M190 S{material_bed_temperature, initial_extruder_nr}
M117 BIBO Printing...

### Actual results

This is the gcode generated by 5.6.0 (installed on 2 Dec 2023):
; Start of BIBO starutp code
M140 S(60, 0)     <------
M104 T0 S175.0
M104 T1 S175
G21  ;metric values
G90  ;absolute positioning
M107  ;start with the fan off
T0
G92 E0
G28
G1 Y0 F1200 E0
T1
G92 E0 ;set extruder 2 position to 0
M109 T0 S175.0   ; now wait for E1  to reach temperature
M109 T1 S175   ; now wait for E2  to reach temperature
M109 T0 S(200.0, 0)  ; now set the correct extruder to the correct initial temperature   <--------
T0
M190 S(60, 0)    <------
M117 BIBO Printing...

I have marked the bad code with <------ on the right.

### Expected results

; Start of BIBO starutp code
M140 S60   <------
M104 T0 S175.0
M104 T1 S175
G21  ;metric values
G90  ;absolute positioning
M107  ;start with the fan off
T0
G92 E0
G28
G1 Y0 F1200 E0
T1
G92 E0 ;set extruder 2 position to 0
M109 T0 S175.0   ; now wait for E1  to reach temperature
M109 T1 S175   ; now wait for E2  to reach temperature
M109 T0 S200.0  ; now set the correct extruder to the correct initial temperature    <-------
T0
M190 S60   <-------
M117 BIBO Printing...

### Add your .zip and screenshots here ⬇️

[bibo2_dual_improved.def.zip](https://github.com/Ultimaker/Cura/files/13537958/bibo2_dual_improved.def.zip)
These are the printer and extruder definitions that have worked in 5.0 and earlier versions.

GregValiant, 2023-12-08:
Looking back I found one of my old posts.
Take a look at #15673 , #16715 (for a BIBO) and #11985 .
On Sept 18 @MariMakes added a ticket to the backlog [CURA-11053](https://ultimaker.atlassian.net/browse/CURA-11053).  It would appear that this still hasn't been addressed.

"initial_extruder_nr" gets replaced but always with a "0" rather than the number of the initial extruder.

I'll mark this as a duplicate report.  It appears that it's on the list of things to fix but it hasn't been repaired yet.

bellzw, 2023-12-08:
Ok.  The bad incorrect replacement of initial_extruder_nr is something I discovered only yesterday.  The error regarding that M109 is a separate item.

GregValiant, 2023-12-09:
I'll remove the duplicate tag.

I think it's time to get a professional involved.

@fieldOfView the OP would like to get something like this to work in the startup gcode to return the proper Initial Temp for the Initial Extruder.
T{initial_extruder_nr} S{material_print_temperature_layer_0}, {initial_extruder_nr}
Is there a solution to this?

bellzw, 2023-12-09:
The exact replacements that we BIBO owners would like to have working are 
M109 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr}
T{initial_extruder_nr}
regardless of whether the initial extruder is T1 or T0.

I appreciate @GregValiant's help with this issue.

casperlamboo, 2024-01-15:
Hi,

Thanks for the bug report. We recently changed some of the behavior regarding start/end extruder/machine gcodes (https://github.com/Ultimaker/Cura/pull/16962). With this change formula's are now accepted with in the curly brackets. With this change apparently an oversight was made regarding the optional extruder argument. It was assumed that only patterns like `{[formula], [extruder_nr]}` are valid, e.g ` M140 S{material_bed_temperature, 0}` will work but ` M140 S{material_bed_temperature, initial_extruder_nr}` won't. It is valid to pass digits, other expressions such as the `initial_extruder_nr` variable were not seen as a valid extruder nr argument

Should be relatively easy to fix by changing this regex to accept any expression after the comma.
https://github.com/Ultimaker/Cura/blob/15b75b7327528ad881aaef9a3c18559b175426ef/plugins/CuraEngineBackend/StartSliceJob.py#L63

bellzw, 2024-01-15:
Thanks for the update, @casperlamboo.  

One comment on the M140: I was wrong to expect M140 S{material_bed_temperature, initial_extruder_nr} to work.  According to fdmprinter.def.json (my BIBO inherits from this definition), material_bed_temperature is NOT settable per extruder.  None of the bed_temperature parameters are settable per extruder, and this makes sense because printers don’t (as far as I know!) have more than one bed.  So, M140 should always be M140 S{material_bed_temperature}, or the like; I’ve corrected my startup code accordingly as the BIBO only has one bed.  As I understand fdmprinter.def.json, resolution of different materials on different extruders having different bed temperatures is by using the maximum of the bed temperatures: "resolve": "max(extruderValues('material_bed_temperature'))". 

There are 2 issues:
1.	 At the end of the startup code, initial_extruder_nr might be replaced by 0, even if the actual initial extruder is 1
2.	 M109 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr} is replaced by M109 T0 S(200, 0).  It should be M109 T0 S200; the () and ,0 don’t belong.  

Recently, I’ve found that issue 1, above, appears may be related to Skirt/Brim Extruder and occurs in 5.6.0 and 5.5.0.  When a print is entirely on E2, and brim is the selected adhesion, these combinations of Build Plate Adhesion Extruder, Skirt/Brim Extruder result in incorrect substitutions: (Not overridden, not overridden) and (E2, Not overridden).  The combinations (E2, E2) and (Not overridden, E2) result in correct substitutions.  It appears that 5.5.0 and 5.6.0 take Skirt/Brim Extruder to be the one to start printing, but there seems to be a disconnect between the initial T1 inserted into the gcode, and the extruder Cura thinks the printing should start with.

Issue 1, above, happens in 5.1.0 in an understandable way if the build plate adhesion extruder is different from the skirt/brim extruder.  Is there a philosophical difference in the way Skirt/Brim Extruder is treated in 5.5.0 and 5.6.0?  In 5.1.0, there is no Not overridden option, Cura interprets Skirt/Brim Extruder to be the extruder that will print the brim, and so it starts with that extruder.  5.1.0 does this when the adhesion extruder is E2 and Skirt/Brim Extruder is E1; Skirt/Brim Extruder seems to override Build Plate Adhesion Extruder, printing starts with T0 doing the brim, and then switches to T1 to do the model.  The actual initial extruder matches the substitution in 5.1.0.

I have attached a pdf with examples of good and bad behavior.
[Issue #17501 examples.pdf](https://github.com/Ultimaker/Cura/files/13942784/Issue.17501.examples.pdf)

closed 2024-01-16 by wawanbreton

MariMakes, 2024-01-19:
Quick update from our side 👋

We released a special Alpha to help with the slicing crashes, but since it's a Work In Progress build it contains the fix that has been proposed. You can download the build yourself here and see if it fixes your problem: https://github.com/Ultimaker/Cura/discussions/18080

Can you let us know if the problem you described in your wonderful PDF is solved or if you would like us to take another look?

bellzw, 2024-01-22:
The issue of M109 T0 S(200, 0) appearing seems to be fixed; I see M109 T1 S200.0 where I should.  However, the issue of initial_extruder_nr being wrong at the end of the startup code remains.

This is what I see when slicing for BIBO2 dual (as distributed with Cura), all extruders are set to Not overridden with the model assigned to E2, no supports, no prime tower,  Normal 25% infill profile, my materials Sunlu White PLA and Green Silk PLA assigned to E1 and E2, respectively:
;Generated with Cura_SteamEngine 5.7.0-alpha.0
T1   <-----Correct
M140 S60
M105
M190 S60
M82 ;absolute extrusion mode
M104 T0 165
M104 T1 165
M109 T0 S200.0    <----- Now fixed in 5.7.0 alpha
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 2mm
T0
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
T0     <--------  WRONG!  Should be T1  Startup code has T{initial_extruder_nr}.  Printing starts with E1 
M117 BIBO Printing...
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:84
;LAYER:0
M107
G0 F3600 X-16.97 Y3.485 Z0.3
;TYPE:SKIRT
G1 F1500 E0

Gcode above starts printing with E1 and never selects E2.  It appears that Cura thinks it's supposed to use E1 for something, despite there not being any mention of E1, except that there is material assigned to the extruder.

When I change only the Skirt/Brim Extruder to E2, I get:
;Generated with Cura_SteamEngine 5.7.0-alpha.0
T1   <---- Still correct
M140 S60
M105
M190 S60
M82 ;absolute extrusion mode
M104 T0 165
M104 T1 165
M109 T1 S215.0   <----- Still correct
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 2mm
T0
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
T1     <----- Correct.  Printing will start with E2, as expected.
M117 BIBO Printing...
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:84
;LAYER:0
M107
G0 F3600 X-16.97 Y3.485 Z0.3
;TYPE:SKIRT
G1 F1500 E0

If I disable E1, I get:
;Generated with Cura_SteamEngine 5.7.0-alpha.0
T1  <---- CORRECT
M140 S60
M105
M190 S60
M82 ;absolute extrusion mode
M104 T0 165
M104 T1 165
M109 T1 S215.0   <---- STILL CORRECT
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 2mm
T0
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
T1    <----- NOW CORRECT.  Printing will start with E2, as expected.
M117 BIBO Printing...
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:84
;LAYER:0
M107
G0 F3600 X-16.97 Y3.485 Z0.3
;TYPE:SKIRT
G1 F1500 E0

The option to set Skirt/Brim extruder disappears, as expected, and now the substitution for initial_extruder_nr is correct.  Apparently, if E1 is enabled and not, as far as I can tell mentioned in any setting, a model entirely on E2 will cause an incorrect substitution of initial_extruder_nr.

I searched for "extruder" in the settings window and enabled every setting that I though referenced an extruder, but did not see anything when slicing to indicate that E1 was referenced.

Let me know if you want me to try other combinations of settings and extruder to which the model is assigned.

casperlamboo, 2024-01-23:
@bellzw, thank your response. The second issue you mention is caused by a feature we implement called multi-material brim (https://github.com/Ultimaker/CuraEngine/pull/1613). To enable multi-material brim the skirt-brim extruder is set to extruder_nr -1 (this is now the default value for this setting). However, there was an oversight made when implementing the feature that, if the extruder_nr was set to -1 the initial extruder nr defaulted to 0, even when the first extruder is not used.

Made a partial fix in https://github.com/Ultimaker/Cura/commit/8af3de2178cee6dc7fb547c75d82b1c4de0bac85.

ansonl, 2024-05-30:
@casperlamboo In the Cura 5.7.1, when I have support enabled and support interface enabled with the support interface set to the second extruder, the `initial_extruder_nr` is set to 1 when the initial layer starts with extruder 0 even though the partial fix mentions that the initial extruder number should default to 0 when no extruder is explicitly set for skirt/brim. Should the initial_extruder_nr be returning 0 instead of 1 in this case according to the fix? 

The Cura preview does not reflect the actual initial extruder number saved in Gcode. I attached a sample project with the S7 set to use support interfaces. The actual gcode extruders being used can be correctly previewed in Prusaslicer Gcode viewer since that viewer labels the extruder colors. 

[image]

For anyone else with this issue, I was able to force the initial_extruder_nr to be 0 by assigning an extruder in Skirt/Brim Extruder. 
This workaround isn't optimal though, because the using different materials in the 2 extruders may need matching material types that would normally be auto assigned without overriding the extruder for the skirt/brim/support.

[image]

[UMS7_p3-repaired.zip](https://github.com/Ultimaker/Cura/files/15506058/UMS7_p3-repaired.zip)

GregValiant, 2024-07-05:
This doesn't appear to have been resolved.
_"we cannot always correctly determine the initial extruder in the frontend without slicing the model."_
The tool callout that appears as a prepend in the data[1] startup section appears to always be correct.  This seems to come down to the sequence of events and at what point in that sequence the replacements are being made in the StartUp Gcode..

Would it be possible to hold off making the StartUp Gcode replacements until there was a chance to go back and look at that first tool number and assign that value to "initial_extruder_nr"?
After that assignment then StartUp Gcode replacements would be made and "initial_extruder_nr" would be real world rather than the default.

PR #18025 (2024-01-16): Allow for expressions in start/end machine/extruder code extruder_nr field
# Description

When using templated string variables in start/end machine/extruder gcode users may enter an optional extruder nr `{expr[, extruder_nr]}`. When providing an `extruder_nr` the expression is evaluated using the extruder stack with the provided extruder_nr. However, this workflow was broken when providing a variable that would evaluate to an extruder nr. E.g. the start code ` M140 S{material_bed_temperature, 0}` will produce the expected result but `M140 S{material_bed_temperature, initial_extruder_nr}` woul [... cut]
commit 8af3de21 (2024-01-23): Fix `getInitialExtruder` code

For the `skirt_brim_extruder_nr` setting it is possible for the setting to be -1; this means that we have no preference for extruder. This allowed us to implement the "multi-material brim". When we were requesting the initial extruder, and this value was set to -1 we were default to the 0-th extruder. However, this was incorrect in the situation where the first extruder is not used.

Fixes #17501