Ultimaker/Cura #17503: Mostruo

carrfel, 2023-12-03:
### Cura Version

UltiMaker Cura 5.6.0

### Operating System

Windows 11

### Printer

Tevo Tornado

### Name abnormal settings

??

### Describe model location

Dragón

### Describe your model

Dragón

### Add your .zip here ⬇️

[Drakon361_ring_LR.zip](https://github.com/Ultimaker/Cura/files/13538434/Drakon361_ring_LR.zip)

5axes, 2023-12-05:
Guass What ?   **Print Thin Walls**

[image]

MariMakes, 2024-01-18:
Hi @carrfel,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 10 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17438

closed 2024-01-18 by MariMakes