Ultimaker/Cura #17505: Cura 5.6.0 OSX-version is BUGGY!

askmurphy, 2023-12-03:
### Cura Version

5.6.0

### Operating System

MacOS Sonoma 14

### Printer

Ultimaker 3

### Reproduction steps

Few problems:
1) MANAGE PRINTER OPTION: doesn't work often, click on the option but nothing happens. After shutdown CURA en started it again it works again but after some time it doesn't.
2) I see often a empty printer slot, see picture below

### Actual results

-

### Expected results

CURA is a bit buggy. the profiles are not very handy too.  After each update of CURA I get my fingers crossed because of the settings. Look at Simplify3D: there  profile/material system is much better!

### Add your .zip and screenshots here ⬇️

[image]

Strykr192, 2024-06-29:
I have the same issue still on 5.7.2 and have had this issue since 5.6

It happens every instance I try to go into any of the settings drop downs after the second time opening one. If I restart the app it'll work for that time until i exit the setting. In my case editing GCode, then upon exit I cant open settings without restarting the app.

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant