Ultimaker/Cura #17520: Bug with Tree suppor creation

rodolfopaulo, 2023-12-04:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro

### Name abnormal settings

No.

### Describe model location

It does nothing

### Describe your model

Yes, I have. It's a box with a name

### Add your .zip here ⬇️

[CE3PRO_Caixa ANABELA.zip](https://github.com/Ultimaker/Cura/files/13549839/CE3PRO_Caixa.ANABELA.zip)

No matter the orientation, the slicer always comes back with a bug. This only happens whem Tree is selected for support. And it needs to be tree.
[CE3PRO_Caixa ANABELA.zip](https://github.com/Ultimaker/Cura/files/13549869/CE3PRO_Caixa.ANABELA.zip)

5axes, 2023-12-05:
I don't think that they need to be absolutly  tree support. but anyway : Tree Support Branch Angle  must be at least 55°

[image]

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)

I'll close this issue since it's resolved, a fix should be available with the upcoming release.

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0