Ultimaker/Cura #17523: i cant slice my stl

Kreituss, 2023-12-05:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

FLSun V400

### Describe your model

just cant slice because have any error anytime

### Add your .zip here ⬇️

[FV_mainbody_bot.zip](https://github.com/Ultimaker/Cura/files/13553860/FV_mainbody_bot.zip)

5axes, 2023-12-05:
No issue on Windows 10
[image]

Kreituss, 2023-12-05:
ok .. fine ... but i have windows 11 not win 10 ...

5axes, 2023-12-05:
I know it's just to give a hint to those looking to solve this problem.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic