Ultimaker/Cura #17524: Excessive CPU usage (Intel macOS)

spamwax, 2023-12-05:
### Cura Version

5.5.0

### Operating System

macOS Ventura 13.6.1

### Printer

Creality CR6

### Reproduction steps

Simply start the app

### Actual results

High CPU usage while idling. Computer fans getting really loud. Rotating and zooming in an empty build plate stutters

### Expected results

Quiet computer and smooth graphics operations

### Add your .zip and screenshots here ⬇️

[cura.log.zip](https://github.com/Ultimaker/Cura/files/13555106/cura.log.zip)

My graphics card and CPU are pretty "powerful" and I can run other slicer and intensive apps easily.

spamwax, 2023-12-05:
Ok, this is getting interesting.
I fully deleted the app and the cura folder in `~/Library/Application Support`
During the first launch of the app if I choose to install an Ultimaker printer (ex Ultimaker S5), I won't have the excessive CPU issue. 
However as soon as I add my printer (Creality CR6), the CPU usage spikes and fans start screaming!
Adding my printer with or without an Ultimaker printer present in the settings causes the same issue!

spamwax, 2023-12-05:
Content of log file *after* adding non-Ultimaker printer: [log.txt](https://gist.github.com/spamwax/1991b1d7df0d28ed2244db331d916252)

GregValiant, 2026-09-29:
I'll close this as stale.

closed 2026-09-29 by GregValiant