Ultimaker/Cura #17526: Slicing fails with tree supports enabled

meismyusername, 2023-12-05:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

VZBot VZ330

### Name abnormal settings

Slicing breaks when enabling tree supports

### Describe model location

2 of model, middle of build plate, position does not seem to change result

### Describe your model

VZBot frame corner brace

### Add your .zip here ⬇️

Hey, not sure what's going on with this one. I'm using pretty much default settings in this profile, and while I can slice with normal supports, tree supports is causing the slicer to error out with "Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker."

[V330_Frame_Brace.zip](https://github.com/Ultimaker/Cura/files/13557048/V330_Frame_Brace.zip)

5axes, 2023-12-05:
No issue 

[image]

deBugger-, 2023-12-09:
Avoid this combination. Tree support and vertical round hole in your Model.
Use normal support instead.
As shown switch from normal to tree support will lead to a cura issue.
[image]
[image]

speedyrudolf, 2024-03-10:
same issue with Cura 5.6.0
no matter what I do, tree supports fail, normal supports work. I have this issue with all models I tried.

closed 2024-04-03 by HellAholic