Ultimaker/Cura #17529: Slicing error, never seen before

CaptScott331, 2023-12-05:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

CR10-S

### Name abnormal settings

no

### Describe model location

no

### Describe your model

not this specific model, but previous models with exact same settings have sliced succesfully

### Add your .zip here ⬇️

[CCR10S_Manual_gearbox_axel_small_parts.zip](https://github.com/Ultimaker/Cura/files/13563437/CCR10S_Manual_gearbox_axel_small_parts.zip)

5axes, 2023-12-05:
Interesting case . The models are simple, but this model has a problem. It needs support and you've defined tree supports. But there's no room to create the support because of the other parts.

[image]

Changing the position of the model or removing the support solves the issue.

[image]

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)

I'll close this issue since it's resolved, a fix should be available with the upcoming release.

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc