Ultimaker/Cura #17530: echec decoupage

tytypsp, 2023-12-05:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

wanhao d 12

### Name abnormal settings

normal

### Describe model location

plat

### Describe your model

non

### Add your .zip here ⬇️

[Great Lion Wall Light.7z-20231204T191034Z-001.zip](https://github.com/Ultimaker/Cura/files/13565006/Great.Lion.Wall.Light.7z-20231204T191034Z-001.zip)
win 11

5axes, 2023-12-05:
As recommended :  
3D PRINT FILE SETTINGS (FOR 0.4 MM NOZZLE)
● Line Width/ Nozzle: 0.4 mm
● First Layer Line Width: 0.48 mm / 120%
● Average Speed: 20mm/s (First Layer) / 50mm/s (All the other layers)
● Recommended Initial Layer Height: 0.20 mm
● Recommended Layer Height: 0.20 mm
● Recommended Perimeters/ Walls: 3.
● Recommended Infill (%): 10% (Gyroid)
● Needs Support: We have printed ours without supports
● Build Plate Adhesion: Skirt.

[image]

No ISSUE .. next

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant