Ultimaker/Cura #17532: Clip SliceFail

Rawlings-T, 2023-12-05:
### Cura Version

5.6.0

### Operating System

Win11

### Printer

Ultimaker S7 Pro

### Name abnormal settings

Non that I am aware of

### Describe model location

Rotated 90 deg to flight

### Describe your model

Sliced successfuly in 5.4.0

### Add your .zip here ⬇️

Project file
[UMS7_Rail Clip V2.zip](https://github.com/Ultimaker/Cura/files/13568350/UMS7_Rail.Clip.V2.zip)

5axes, 2023-12-05:
Windows 10 No Issue .. Windows 11 ?
[image]

Rawlings-T, 2023-12-05:
Correct Win11

GregValiant, 2023-12-05:
I've noticed that myself.  I read an interesting article that _might_ have some bearing on this.  It was about the tendency of computer manufacturers to over-clock their processors trying to squeeze out more performance.  Since newer machines are more likely to be running Windows 11 the article got me thinking if the problem might have some basis in that over-clocking.  It's a subject that I know nothing about, but it was an interesting take on something that might be going on.

5axes, 2023-12-06:
Duplicate ?
https://github.com/Ultimaker/Cura/issues/17538
https://github.com/Ultimaker/Cura/issues/17530
https://github.com/Ultimaker/Cura/issues/17526
https://github.com/Ultimaker/Cura/issues/17523
https://github.com/Ultimaker/Cura/issues/17509

Rawlings-T, 2023-12-06:
Unsure if those are duplicates or not. Not created by me but they seem to all be relatively thin parts facing slicing issues with Win11 on cura 5.5 or 5.6. Could be same issue in cura.

closed 2024-04-03 by HellAholic