Ultimaker/Cura #17536: Cooling does not work

dzimme, 2023-12-06:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Anycubic Viper

### Reproduction steps

G-Code store on SD
Start Printing

### Actual results

Cooler every time by 0%

### Expected results

Start cooling by 40% , on Layer 2 80%

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-12-06:
Thanks for the report.
From top to bottom your settings should be:
Fan Speed.............80
Regular...................80
Maximum..............80
Threshold..............10     (this won't come into play because "Max" and "Regular" are the same)
Initial Fan Speed..40
Regular Fan Speed at Height (--)     (using "layers" will override "height" so this one will gray out)
Regular Fan Speed at Layer 2

You should see "M106 S102" at Gcode ";LAYER:0" and "M106 S204" at Gcode ";LAYER:1".

I'll remove the bug label as this is a settings issue and I'll go ahead and close this.  If you feel that is premature let me know.

closed 2023-12-06 by GregValiant

dzimme, 2023-12-14:
Dear Mr. Valiant, I have testet your suggested parameters, with no success.
The last release for a well cooler function is Cura 5.3.0. All the further 
releases supports not the cooling funktion by Anycubic Viper.
 
Now I work with Cura 5.3.0 , and have a hope that this problem can be 
solved in new releases.
This software is free , and I be thankful for your work. I wish you and 
your team a peaceful time over the change of the Year. Thanks for all.
Dieter Zimmermann

 
 
 
 

<https://github.com/Ultimaker/Cura/issues/17536#issuecomment-1843050344> , 
[ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage", 
"https://github.com/Ultimaker/Cura/issues/17536#issuecomment-1843050344", 
"url": 
"https://github.com/Ultimaker/Cura/issues/17536#issuecomment-1843050344", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"https://github.com" } } ]

GregValiant, 2023-12-14:
Open a gcode file you created with 5.4, 5.5, or 5.6 with a text editor.  Search for "M106" which is the command to control the fan speed.  "M107" turns the fan off.

With a Vyper as the active printer, I don't have any trouble having the commands show up in the gcode.  Typically, there will be one command at the layer I indicated the fan should start.  At 80% fan speed the command is "M106 S204".  If the command is there but is "M106 S0.8" then the settings "Fan scale 0 to 1" has been checked.  It is in the Printer Settings and is specifically for some RepRap machines.

No other Vyper owners are having this problem.  I cannot duplicate this problem.  I just don't see a problem here with Cura 5.4, 5.5, or 5.6. 

If you wish to post a gcode file that you created with 5.6 and a project file from 5.6 I will check them.  This simply does not look like it is a Cura problem.