Ultimaker/Cura #17538: Slicing issue with Cura 5.6.0

ferorga, 2023-12-06:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Kingroon KP3S Pro

### Name abnormal settings

I've changed many. It does slice with default profiles

### Describe your model

It is just a box created from fusion360 parametric model. It does slice with Cura 5.3.1

### Add your .zip here ⬇️

[KK_skadis_measure_tape_CRASH.zip](https://github.com/Ultimaker/Cura/files/13583181/KK_skadis_measure_tape_CRASH.zip)

GregValiant, 2023-12-06:
Thanks for the report.
I didn't have any trouble slicing this on my Win10 PC.
This another report involving slicing problems on Windows 11.  It would be helpful if you could provide more information about the computer that is having the problem.  In particular the processor and graphics card info.  They are available at "Start/ Settings / System / About...".

I'll mark this as a duplicate and the Cura team will take a look.

5axes, 2023-12-06:
yes no issue on Win 10 ...
[image]
Using Cura on Win 11 seems to generate quite a few problems

MariMakes, 2024-01-08:
Hi @ferorga,

Sorry, it took us a while to get back to you 😞

I'm on Windows 11 and I took a look at it in Cura 5.6 and I've been able to slice it, so I don't believe that it's a windows 11 issue but something else related to your set-up 🤔

[image]

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

Sasch600xt, 2024-02-01:
Same issue I ran into today. Also support was set to "tree". After I changed it to "normal" I could slice like normal.
So maybe an issue with tree support?

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic