Ultimaker/Cura #17541: Set print temperature for One at a Time prints

rspangit, 2023-12-06:
### Is your feature request related to a problem?

When evaluating a new material, I run a series of small models while adjusting settings for each model.  The easiest way is to print one at a time with per-model settings.  I can do this for print speed and flow, but Cura does not let me set print *temperature* per model.

Most recently, I ran into this while experimenting with ColorFabb LW-PLA.  I printed a large number of hollow test cubes (<5 min per print), and couldn't just run one job with a dozen cubes spanning the whole 200C - 280C range.

### Describe the solution you'd like

I'd like to be able to set print temperature per model for one-at-a-time prints, the same way I can set flow or print speed.

### Describe alternatives you've considered

Hand-editing the gcode does not work well, because the models are seemingly printed in random order.  So I wouldn't be able to get a run with temperature increasing back-to-front left-to-right.

### Affected users and/or printers

This would benefit anyone who runs test prints at varying temperatures.  Which is probably darn near everyone who doesn't just stick with the stock profiles.

GregValiant, 2023-12-07:
This could be done in post process.
A problem is that a script like this might be useful to you, but most folks can get by with ChangeAtZ so I think this is a niche thing. 
 You need a custom script that would insert temperature changes at every "LAYER:0" and list which models were printed at what temperature.  You would also need a numbered sketch of the build plate layout showing the printing order.  I think that would be the only way to cross-reference the temperatures to the actual prints.

5axes, 2023-12-07:
Duplicate for the script solution : https://github.com/Ultimaker/Cura/issues/8555

And if it's just to test the material I think you can do that with the 4.20 , but not absolutly sure need to check.

GregValiant, 2023-12-08:
At this point in time (and probably for the foreseeable future) I don't see any way around a hand written cartoon.  You pull the numbers from the Cura preview to get the print order and fill in the balloons with the model names/numbers.
[image]

The post processing script adds temperatures and pulls the mesh names that are within each layer 0 and writes that list into the gcode something like this.  This gives you the the print order and the temperature that was assigned to each model.
;...Order..........Model Name.........................Temp
; ....1)........Cane Rubber.stl(2).......................200
; ....2)........Cane Rubber.stl(1) ......................210
; ....3)........50mm Cooling Duct V4.stl.......215
; ....4)........Cane Rubber.stl............................222
; ....5)........X-Y Ref Block.stl(3)......................225
; ....6)........X-Y Ref Block.stl(2)......................227
You match up the names from that list to the names in the balloons on the cartoon and you now know which model printed at what temperature.  You just need to remember to mark them before you rip them apart.
Now that this is all designed I'll go back to watching some American football.

rspangit, 2023-12-08:
I don't see mesh temperature delta in the 5.6.0 release.  I gather that's unique to the 4.20 tree.

I don't see Ultimaker S7 support in the 4.20 tree:
https://github.com/smartavionics/Cura/tree/master/resources/definitions

And even if it were there, the whole point of running experiments before production is to test how the material behaves with _only the variables under test_ changing (that is, the print temperature).  Not all the other parameters which behave differently between 4.20 and 5.6.0 (gradual flow, etc).  It would produce a result, yes, but not one which necessarily correlates with how the current version of Cura works.

So, since we're talking about American football, that's about as useful to me as Taylor Swift is to the Chiefs.  Sure, _one_ of the players is scoring, but that doesn't do much for the team's score.

However, this discussion led me to look more closely at how Cura does one-at-a-time printing.  The models appear to print last to first in the object list.  It's just that they're _arranged on the build plate_ in a random order.  So the useful technique is to drag the objects into the order they're printed rather than relying on how Cura places them by default.  Verify using the preview that is indeed the order they'll be printed in.  And _then_ a script which changes temperature at each `LAYER:0` will do the right thing and consecutive locations on the build plate will get printed in increasing temperature.

That is: don't _write down_ the funky order that Cura wants to print objects by default.  Just _drag them into a more sane order_.

(Why do they always print last to first instead of first to last?  That's another mystery.  But as long as it's consistent, it's a feature.  I can work with that.)

5axes, 2023-12-08:
Very eager to see what our Patrick Mahomes will do for our team.