Ultimaker/Cura #17542: Output gcode causes extruders to lock up

DrMeo, 2023-12-06:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality CR-10 / Creality CR-10S S5

### Reproduction steps

Cura shows no errors when slicing. 
 
When slicing the base.stl from https://www.thingiverse.com/thing:5214344 I have two different extruders that lock up and will not extrude forward at layer ~5-6 ( 1.5mm Z height ) every time I try to print these files when sliced with Cura.

The lock up happens in the exact same place every time. This includes both with Octoprint from a RasperryPi, and directly from SD card. 

Extruder lockup means no clicking from skipping steps, no grinding of the filament on the extruder gears, and you can not move the extruder in the forward direction even when grabbing the stepper motor shaft. 

I can provide gcode files for the files sliced in cura that fail, as well as the working files from Prusa slicer for comparison if need. 

### Actual results

Extruder locks up at layer 5-6 and will not move.

Extruder lockup means no clicking from skipping steps, no grinding of the filament on the extruder gears, and you can not move the extruder in the forward direction even when grabbing the stepper motor shaft with pliers. If you cancel the print and immediately set your hot end temps to what they were while printing the extruder will extrude like normal immediately after you manually perform any retraction. All other files that are sliced will print fine.

This happens on two separate machines, with two different extruder motors, and two different main boards and stepper drivers. The CR-10S S5 has a Creality 4.x "Ramps" main board, and the CR10 has an original "Melzi" board.  

### Expected results

Cura's gcode should print and extrude like every other print file has for the last several hundred prints sliced with Cura with these exact settings on these machines. 

When the file is sliced with Prusa slicer 2.6.0 they print normally with no extruder lock up on both printers. 

### Add your .zip and screenshots here ⬇️

[CCR10_BASE.zip](https://github.com/Ultimaker/Cura/files/13591541/CCR10_BASE.zip)

GregValiant, 2023-12-06:
Thanks for the report.
This is the first report of this behavior that I've heard.  I haven't had any trouble printing anything with my Ender.
PrusaSlicer Gcodes are in "relative extrusion" mode.  I don't see how that could be the issue but for a test you could try changing to "relative extrusion" (it's in the Special Modes section).
I just don't see anything in the Cura gcodes that could cause a printer to "lock up".  There certainly isn't anything odd in the gcode I generated from your project file.

It's odd that whatever is happening is affecting both printers and that no one else has mentioned this.  5.4 was released 5 months ago and there are enough CR-10 owners out there using Cura that someone else should be having the same issue.  I've been doing triage here for a couple of years and as I say, this is the first report of this "extruders locking up" problem that I've seen.

I'll leave the bug label on this and the Cura team will take a look.  If the problem can be duplicated then the investigation can continue, but I can't duplicate it myself.  My printer is running now and is up to layer 576 of the current print.

Don't put blinders on as there is a good chance (the "both printers" thing not withstanding) that this is a local problem.

DrMeo, 2023-12-07:
It's also the first time I've seen this happen as well, usually extruder issues are either something too tight and it sits skipping steps, or just grinds the filament on the drive gears. I've never seen a functional extruder just plain stop and refuse to go forward. I just get lucky and find odd corner cases... lucky me. 

That being said, I have an update: 

The relative extrusion did not help, still locked the extruder at Z1.5mm in the exact same area. 

BUT I noticed that Prusa had flipped the orientation 180 degrees when I put the flat base on the bed. When Cura slices at the exact same settings that fail when the hinges are towards the back, but with the part rotated 180 degrees so the hinge side of the model is facing the front of the printer instead, it prints with no issues. 

So something in the orientation of the part is causing Cura to output gcode that both of my extruders don't like, without the engine catching anything.

DrMeo, 2023-12-07:
> When the extruders stop - do the other motors continue to function and the print simply continues with no extrusions? Have you noted the hot end temperature when the extrusion stops? Has this problem occurred with any other models? Do you happen to be running "stock" firmware in both of those printers? Is the firmware the same in both printers?

When the extruders stop it's exactly that, they stop and do not try to extrude any more. No grinding of the filament or anything, just a dead stop. Not even any sounds or jerking of the motor shaft like the stepper motor is skipping steps. The rest of the print keeps going like it was still extruding with no issues other than zero extrusion. I can release pressure from the drive gear and push filament by hand and extrude plastic onto the part if the nozzle is still close enough to the part. 

Hot end stays stable @200C according to the temperature log graph in Octoprint. I don't see anything in the terminal log that would indicate the firmware crashed or had any issues at all either. Just normal SEND and [OK] responses. 

This is the only model that's done this, hence why I reported it. It seems to be a strange edge case that may not have a huge impact on a lot of users, but might point to some other ways that Cura can do unwanted things even though the engine thinks slicing went fine. I have finished prints yesterday after reporting that were sliced with the exact same setting and printed fine, I didn't even close the Cura window after slicing, just deleted the other models and imported new and sliced again. 

Firmware is TH3D 2.74a on the CR10S S5 with the 4.x "Ramps" silent stepper board, and 2.71 on the noisy old CR10 Melzi board. 

I didn't see anything glaringly obvious in the gcode myself, other than noticing this issue _seems_ to maybe happen after Cura does rapid stitching to fill in between walls where it either can't or won't print an extra wall to compensate.

GregValiant, 2023-12-07:
Ah Grasshopper, but you don't have my fancy macros to analyze gcodes in Excel and AutoCad.

I was absolutely 100% right, there is nothing in the Gcode to **_STOP_** the extruder and in fact it never stopped.  It was just turning so slowly that it made a tortoise like like a speed demon.

Line 15918 has a prime and I'll post the snippet here.  Look at the E values.
 ;TYPE:SKIN
G1 F2700 E973.2293 ---> prime after travel
G1 F1100 X148.442 Y243.577 E973.22934
G1 X148.584 Y243.577 E973.22946
G1 X148.413 Y243.747 E973.22966
G1 X148.413 Y243.888 E973.22977
G1 X148.725 Y243.577 E973.23014
G1 X148.867 Y243.577 E973.23026
G1 X148.413 Y244.03 E973.23079
G1 X148.413 Y244.171 E973.23091
G1 X149.008 Y243.577 E973.23161
G1 X149.15 Y243.577 E973.23173
G1 X148.413 Y244.313 E973.23259
G1 X148.413 Y244.454 E973.23271
G1 X149.291 Y243.577 E973.23374
The delta E in those 12 lines is .00428mm.  At 0.2 LH and 0.4 LW that's a total of 0.13mm of extrusion and I can see that there is certainly more length there than that.
The entire section of Skin from line 15918 to line 20374 is at 0.01 line width.  Talk about under-extrusion!!!
This happens again in Layer 55 at lines 124460 thru line 136489 and once again it's a Skin section as is the aberration in Layer 67.
There are a lot of line segments in the gcode that are shorter than .00000001mm.  That has me thinking that your "Maximum Resolution" might be set too low.
This is from AutoCad.  The line width of the close-together lines is 0.10mm.  That's way too close and so it's possible that your "Minimum Line Width" settings were way off.
[image]

So based on the above observations and my painstaking analysis of the relevant data, I'm declaring this might be a bug or it might not.  I'm positive about that.

@DrMeo if you can recreate the situation in Cura that produced that gcode file, then use the "File | Save Project" command and post the zipped project file here it might shed some light on this.

DrMeo, 2023-12-07:
The original project .3mf file is in the gcode.zip file I have attached to the original report. 

It should be reproducible for both printers, as I sliced the file multiple times and tried to print several times on each printer. As an added test I loaded the model, sliced, and exported the project with my normal settings and also rotated 180 degrees - which the rotated part DOES fully print successfully.   I also included a slice of the original settings + disabling ironing, even though ironing is generally on on all of my prints and while a mark on the filament moves slowly over time, does still noticeably move over a short period of time. Perhaps it's something to do with the orientation of some moves + ironing? 

I normally print a lot of figures and boxes on these machines, so ironing doesn't much affect the figures, and is left on at all times, but makes significant improvement on the boxes. Plus if it was ironing moving extremely slowly the rotated part should exhibit the same behavior, in theory anyways.

Resolution is the standard generic PLA settings - Minimum line width is 0.34, and maximum resolution is 0.25 with maximum deviation of 0.025. Print thin walls is disabled. 

Here is the set of slices and project files  to see if the super thin lines are reproducible again: 
[slicedprojects.zip](https://github.com/Ultimaker/Cura/files/13606164/slicedprojects.zip)

GregValiant, 2023-12-07:
You had "Ironing" enabled at 0.1 line width and 10% flow.  Those are the numbers I saw in the analysis so this was a settings issue.
I'll go ahead and close this.  The printer was doing what it as told to by the gcode and that was to barely extrude on the top skin.  Let me know if you feel that was premature.

closed 2023-12-10 by GregValiant