Ultimaker/Cura #17543: File crashed while slicing

CIPuyleart, 2023-12-06:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ultimaker S3

### Describe your model

STL exported from Shapr3D

### Add your .zip here ⬇️

[UMS3_Console No Top Holes.zip](https://github.com/Ultimaker/Cura/files/13591827/UMS3_Console.No.Top.Holes.zip)
this is slicing and printing fine in Cura 5.4.0

5axes, 2023-12-06:
No issue on Windows 10

[image]

Certainly  the same issue as every report concerning Cura 5.6 and Windows 11

CIPuyleart, 2023-12-06:
the only extra step done on this file was rotating it 180 degrees about Y axis prior to slicing. Just tested with it in original orientation and still crashes.

5axes, 2023-12-06:
ok same issue not related to windows 11

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic