Ultimaker/Cura #17545: failed to add printer

RJH-UK, 2023-12-07:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

GEEETECH A20

### Reproduction steps

Clicked on add printer none ultimaker printer, none networked , geeetech, a20, add

### Actual results

cura closed down

### Expected results

added printer a20

### Add your .zip and screenshots here ⬇️

nothing else to add

MagicTF, 2023-12-07:
this also happening to me, all versions following 5.2.2 are crashing when first setup is run and Geeetech A20 is selected. Workaround is to select custom FFF printer and align all values for the printer and the extruder. But I do not know whether this is healty enough as I might be missing certain definition from the original A20-config files.

MagicTF, 2023-12-07:
just did a file comparison for Geeetech_A20*.json between V5.2.2 and V5.6.0.
The result was that in the C:/UltiMaker Cura 5.6.0/share/cura/resource/extruders the file for the A20 was missing. So I copied "geeetech_A20_1.def.json" from the V5.2.2 installation to V5.6.0 and I could install the A20-printer without crash. For those who do no not have a V5.2.2 installation I attach the file.
[geeetech_A20_1.def.json](https://github.com/Ultimaker/Cura/files/13600795/geeetech_A20_1.def.json)

content of the file:
------snip--------------
{
    "version": 2,
    "name": "Extruder 1",
    "inherits": "fdmextruder",
    "metadata": {
        "machine": "geeetech_A20",
        "position": "0"
    },

    "overrides": {
        "extruder_nr": { "default_value": 0        },
        "machine_nozzle_size": { "default_value": 0.4 },
        "material_diameter": { "default_value": 1.75 }

    }
}

Asterchades, 2023-12-08:
So near as I can figure this is an unfortunate oversight made during a change about 9 months ago, just prior to the launch of 5.3.0. Several Geeetech definitions, including the A20, were reverted to earlier versions and _most_ of their previous Extruder definitions were restored - but not the base A20 extruder.

I'm going to ask the dev responsible for the reversion to take another look at it. The commit references an engine issue so I'd rather not blindly request the definition just be readded in case there's a specific reason it's missing.

In the meantime, however, your proposed solution is what I would do. It _shouldn't_ cause any problems, but there could be something more here so I'm reluctant to say that it won't. For people who just want a pre-composed file to download I would recommend grabbing a copy of the old 5.2 file from here: https://github.com/Ultimaker/Cura/blob/5.2/resources/extruders/geeetech_A20_1.def.json

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant