Ultimaker/Cura #17549: Slicing failure

dreeco, 2023-12-07:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Ender 3 S1 pro

### Describe your model

simple headset

### Add your .zip here ⬇️

[serretete_1-9-.zip](https://github.com/Ultimaker/Cura/files/13600317/serretete_1-9-.zip)

5axes, 2023-12-07:
An other Tree Support issue.

[image]

As this part is not adapted to the tree support, the creation of the supports cannot take place and the cutting process fails.

GregValiant, 2023-12-07:
Now we don't know because it sliced fine for me with Tree Supports turned on.  This is on my Win 10 Pro laptop.

dreeco, 2023-12-07:
Thank you for your answer.
When I disable support it works! Thanks !

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)

I'll close this issue since it's resolved, a fix should be available with the upcoming release.

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0