Ultimaker/Cura #17552: uno bottom

Jamlz23, 2023-12-07:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

Artillery sidewinder x2

### Name abnormal settings

no

### Describe model location

si

### Describe your model

Si lo eh impreso antes y manejado correctamente

### Add your .zip here ⬇️

[ASX1_bottom.zip](https://github.com/Ultimaker/Cura/files/13603415/ASX1_bottom.zip)

5axes, 2023-12-07:
No issue on Windows 10 ... Could be the Windows 11 problem.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic