Ultimaker/Cura #17554: Printing an object that is placed above Zero Z (drop down model is unchecked) always start printing from Zero Z

StmLord, 2023-12-07:
### Cura Version

5.5

### Operating System

Mac OS Monterey

### Printer

Custom FFF

### Reproduction steps

1) place an object on bed
2) uncheck drop down model and lift it from bed
3) slice 

### Actual results

both preview and real printing starts at Zero Z axis.

### Expected results

slicing should respect Z height of an Object. 

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2023-12-07:
Thanks for the report.
There is a setting bug involved here.

- Disable "Generate Support"
- Go to the Mesh Fixes section and disable "Remove empty first layers".
- Turn supports back on (if you need them).

The bug is that if the setting is going to impact the slice, that it should be visible.
I'll mark this as a duplicate for the Cura team.
[image]

StmLord, 2023-12-19:
Yes, thats it, I had to enable the view of "remove empty first layers" setting to disable it and that did the trick.
Thanks

GregValiant, 2026-09-29:
Looks like this got fixed.  I'll close this report.

closed 2026-09-29 by GregValiant