Ultimaker/Cura #17555: Slicing takes 4e4 on Big Models

Cascadier1, 2023-12-08:
### Cura Version

6.5

### Operating System

Win11

### Printer

No Matter

### Reproduction steps

Trying to slice "EifelTower Fixed" Model from Thingiverse, scaled @250%. No Error but it takes 4e4.
Canceled after 2,5 Hours. No Problem with an older 4.x Version. 

### Actual results

Still Waiting

### Expected results

I would get a sliced Model

### Add your .zip and screenshots here ⬇️

[EN3MAX_EiffelTower_fixed.zip](https://github.com/Ultimaker/Cura/files/13610204/EN3MAX_EiffelTower_fixed.zip)

5axes, 2023-12-08:
It s not specific to "big" model but it's an issue detected on model with a lot of cutting area, like your eiffel  tower or a voronoï model. Cura 5.5 & 5.6 takes more time to slice a model . Most of the times it's a few percents but for these types of geometry can be much , much , much more longer 😢

5axes, 2023-12-08:
It's a problem that appeared with version 5.4 ... why? you'll have to ask the developers. My answer was only meant to give the beginning of an analysis, I know it doesn't solve anything. I do hope that this problem will be taken seriously for future versions.

GregValiant, 2026-10-01:
I'll go ahead and close this as the problem has been addressed.  The model sliced in 56 seconds in Cura 5.13.0.

closed 2026-10-01 by GregValiant