Ultimaker/Cura #17558: Slicing error

JVPFowler, 2023-12-08:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ultimaker 3 extended

### Reproduction steps

Trying to merge two components. Created the main component (key fob body ) with a cavity carved out for the lettering. Created the lettering to fit the cavity. Initially tried to merge the two so they'd print in different colours. This didn't slice well with the final layer being in the body colour not the letter colour.  Redrew the item and tried to merge them but they stayed separate so ended up manually merging the two components. Sketchup was used to create the 3D model. The lettering was downloaded separately from the body as STL files before opening in Cura.
[image]

### Actual results

When sliced, the software appears to ignore the fact there's a cavity in the surface of the key fob body for the lettering.  When printing, the body material (blue) would print over the highlighted letter material (white) so I had to manually raise the lettering above the surface by three layers.  Not the ideal end product as I wanted a smooth surface with the lettering inset and flush with the surface.

### Expected results

I've attached both the downloaded PNG file showing the two components and the resulting Ultimaker project file.

### Add your .zip and screenshots here ⬇️

[ED Key fob.zip](https://github.com/Ultimaker/Cura/files/13612786/ED.Key.fob.zip)

JVPFowler, 2023-12-08:
Hi Greg. Thanks for your quick response.  Cura didn't report errors in the model as it usually does if there's a fault. I am, by my own admission, very much a novice at 3d printing, so I'm not sure what you mean by flipped normals? ...and I'm not sure how to cure the problem...

GregValiant, 2023-12-08:
Any surface has two sides.  If the surface is one of the ones enclosing a volume (like one of our models) then all of the "inside faces" of the surfaces must face the inside and the "outside faces" face outside.  (actually, if they are all 100% facing the wrong way that will work too).
In your model the red surfaces were facing outboard and they should have been facing inboard.  That means that they were "flipped" (or "reversed").  Cura saw the flipped faces and so added material onto them because it thought it was the inside.

Non-watertight models have open seams.  If you were to try to fill them with water it would leak out and hence the name.  The problem they pose is that as the slicer goes around a corner that is an open seam, it goes inside the model and get confused as to where the outside ends and the inside begins.

You just need to be careful when designing things.  If you use something like Blender it should have tools to allow you to flip surfaces.  I don't know Blender but some other apps are able to do that.  Look for color differences between one side of a surface and the other.  Sometimes there will be a little point that indicates the "outside" of a surface.  Each app is a little different in that regard.

MS 3D Builder is available in the Microsoft store.  It's pretty good at making repairs and at altering STL files.  What it isn't  -- is intuitive.  I've gotten better at it but my CAD background doesn't help a lot.