Ultimaker/Cura #17562: Grid infill incorrectly rendered

etkaar, 2023-12-08:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

FLSUN Q5

### Reproduction steps

Any ideas what the reason for that could be?

[image]
[image]

### Actual results

See above

### Expected results

See above

### Add your .zip and screenshots here ⬇️

See above

GregValiant, 2023-12-08:
Thanks for the report.
I've seen this myself but now when I want to reproduce it I can't.  I wanted to check to see if it makes it into the gcode.

The Cura team will take a look.

etkaar, 2023-12-08:
Thank you! Please find attached the 3D file and the printing profile
[3DFileAndPrintingProfile.zip](https://github.com/Ultimaker/Cura/files/13621625/3DFileAndPrintingProfile.zip)
, maybe that helps.

GregValiant, 2023-12-09:
I can't reproduce it with your file either.  It could be a glitch in the preview.
This is with an FL printer and your file and settings.
[image]

GregValiant, 2026-10-01:
I'll close this.

closed 2026-10-01 by GregValiant