Ultimaker/Cura #17563: Pain On Supports

RetArsonFD, 2023-12-08:
### Is your feature request related to a problem?

I can't stand the "Custom Supports".  Most of the time, in order to get supports where I need them, I need to enable all supports, then individually block out areas I do not need supported.  Way too much work, when other slicers already have paint on supports where you need them.

### Describe the solution you'd like

I'd like to see paint on supports added as a feature.

### Describe alternatives you've considered

Not sure.

### Affected users and/or printers

I hope everyone will benefit in some way.

Asterchades, 2023-12-09:
This has been asked for more than once since PrusaSlicer incorporated them. Honestly I'm not sure what the current state of the request is as there's been no mention on a forum I can access.

Duplicate of #14648.

closed 2023-12-09 by Asterchades

RetArsonFD, 2023-12-09:
I figured it probably had already been requested, but hoping that another request MIGHT persuade the programmers a little more to incorporate the feature.  THANKS !!!!!!!