Ultimaker/Cura #17565: Slicing failed for this project

dobbleg1000, 2023-12-08:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3 S1 plus

### Describe your model

Model of the one ring for thingiverse

### Add your .zip here ⬇️

[CE3S1_small1.zip](https://github.com/Ultimaker/Cura/files/13621402/CE3S1_small1.zip)

GregValiant, 2023-12-09:
Thanks for the report.
4th one in a row that won't slice in Win11 but does in Win 10.
I'll mark this as a duplicate.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic