Ultimaker/Cura #17566: nothing will slice

Nightsfury2677, 2023-12-08:
### Cura Version

5.6.0 or 5.4.0 nothing will slice on either

### Operating System

windows 11 

### Printer

anycubic vyper

### Name abnormal settings

none

### Describe model location

centered

### Describe your model

mack truck sliced before printed but crashed (due to blender) 

### Add your .zip here ⬇️

[1968_Mack_DM-600_DETAILED_v23.zip](https://github.com/Ultimaker/Cura/files/13621452/1968_Mack_DM-600_DETAILED_v23.zip)

Nightsfury2677, 2023-12-25:
the issue is WINDOWS 11 related droping back to 10 or compatability to 8 is the only way it works for me now

GregValiant, 2026-10-01:
5.5 and 5.6 had problems with certain types of models.  Most of that was fixed in 5.7.0 and since then, there have been numerous improvements.  The current version is 5.13.0 so it is time to update.

closed 2026-10-01 by GregValiant