Ultimaker/Cura #17567: Imported  a file from prior version of cura, and sliced it.

servant74, 2023-12-08:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ultimaker 3

### Name abnormal settings

None.

### Describe model location

Centered on build plate

### Describe your model

2 color medal for girl scouts, wife is developing it for local girl scouts council

### Add your .zip here ⬇️

[UM3_ThinMintSprint 2 color-2024.zip](https://github.com/Ultimaker/Cura/files/13621628/UM3_ThinMintSprint.2.color-2024.zip)

GregValiant, 2023-12-09:
Thanks for the report.
This is another one that will slice in Win10 but (apparently) not in Win11.  I'll mark this as a duplicate and @5axes will take note.

You can try going back to Cura 5.2.2 to slice this yourself.  I don't have a Win 11 machine to test on, but there seems to be a definite problem between Win11 and Cura 5.6.

Here is the gcode file from the slice I did with your settings (at least you can do a test print).
(A little known fact is that I was an Honorary Girl Scout of troop 344.  I insisted because my mother was the troop leader and I had two sisters in it so they dragged me all over the place just to set up the camping equipment.)
[UM3_ThinMintSprint 2 color-2024.zip](https://github.com/Ultimaker/Cura/files/13624349/UM3_ThinMintSprint.2.color-2024.zip)

servant74, 2023-12-09:
Side note: Not related to the problem.

My daughter wanted to get into Horses while in GIrl Scouts at one of their
camps when we lived in Houston,
since it was a couple of hours each way my wife stayed with her, then
started getting active there.  Eventually they lived
on Camp on the weekends it seemed.  Eventually my wife was one of their
riding directors.

That went on for a couple of years.  My wife lost interest in her job at a
pre-school and our kids were older, and my job was
no fun, so she started looking for another job.  She found one in TN on a
lark she applied to be Equestrian Manager for
a Girl Scout Camp.  That went well so we moved to TN onto the Camp.  15
years later she retires, and we stay in the area
with my daughter living just down the street (daughter works as a teacher,
and helps at the Camp just 30 minutes away).

Our daughter gets interested in 3D Printing (she has 4 Ultimaker 2's) and
so do I (UM3).  She prints for science/math STEM
teaching needs, and I play.  Wife got pulled back in to write grants
(something she never did before) and now she is doing
the medals like you see for the Scouts like you noticed in the project
file.

So this is part of staying busy in our retirement and my wife

GregValiant, 2023-12-09:
Keeping the mind active is really important as we get older.  This kit printer I have is good enough for what I do.  Cura is highly customizable and so I've been writing my own post processors.  I've learned a lot here doing the triage but now it's getting to be same thing after same thing with this "No...I won't slice this" bug.
PS  My wife likes the mints but I'm a fan of dunking the shortbreads in my morning coffee.

5axes, 2023-12-10:
Yes no issue for me on W10 added to the list.

[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic