Ultimaker/Cura #17568: Slower wall speed than configured (spiralized outer contour)

tommyhp2, 2023-12-09:
### Cura Version

5.6 and 5.5

### Operating System

Windows 10

### Printer

Creality Ender 5 Plus

### Reproduction steps

Here's a 2D sketch of the model:

[image]

On the straights, it's about 42 mm.  The 3D model is 0.4 mm shell all around (including bottom)  The configured print speeds are as follow:

80 mm/s (including infill)
66 mm/s all walls
60 mm/s initial and top/bottom

0.2 layer height
0.4 nozzle and line width
0.8 wall
0.4 top/bottom (forcing spiralized outer contour to print 1 bottom layer instead of 3 at default 0.8 )

### Actual results

Initial and 1 bottom layer prints at configured speed as expected.  The wall speed prints at configured speed for about 2 layers then drops to 40 mm/s or less on subsequent layers.  (video too big to upload to github.)  Here's the link to video showing actual slower wall speed:  https://youtu.be/gsdBP7_FfSM  (video still uploading at time of bug report)

### Expected results

66 mm/s wall speed, especially the straights, on all layers after initial & bottom.

### Add your .zip and screenshots here ⬇️

[tst_z_bnd2_4_70.zip](https://github.com/Ultimaker/Cura/files/13621844/tst_z_bnd2_4_70.zip)

tommyhp2, 2023-12-09:
Because of the incorrect wall speed, the total print time came out way longer than expected:

1. The initial and 1 bottom layers took about 4.5 minutes each
2. The remaining layers should be: ( 70 mm [total height] - 0.4 [for initial and 1 bottom] ) / 0.2 = 348
3. Time to print each side on subsequent layers per configured wall speed should be about 1.25 second at most.  Each layer should take about 5 seconds to complete.
4. Total time of print should be:  4.5 * 2 * 60 + 5 * 348 = 2280 seconds = 38 minutes.  Because of the slower wall speed, it ended being about 69 minutes.

I did some additional testing regarding 'wall' configurations.  Courtesy of [Tyler Foust's US map in 3 lines](https://twistedsifter.com/2020/12/map-of-the-usa-in-3-lines-by-tyler-foust/),  the configured seam is 'back left' which should be most northwest of US, ie Washington.  Regardless of what the seam location and hiding preference, it always starts in southern California.
[us_map_3_lines.zip](https://github.com/Ultimaker/Cura/files/13622420/us_map_3_lines.zip)

Asterchades, 2023-12-09:
You're configured for a Minimum Layer Time of 10 seconds, with a Minimum Speed of 10mm/s. It's not in the least bit unexpected that the time is nearly double what you're expecting if you've calculated that the layers should be taking 5 seconds (or less) at the correct speed. Reducing the minimum time or increasing the minimum speed will be required if you want to maintain your expected wall speed.

In regards to your other issue, the seam in the project as provided is set to the Front Left - ie California. Setting it to Rear Left (Washington) without changing any other setting works as expected, as does manually entering co-ordinates to use (at least under 5.6). So I'm not quite sure what you're experiencing there or why you're experiencing it. There are known issues with seam locations but this does not appear to be one of them.

tommyhp2, 2023-12-09:
> You're configured for a Minimum Layer Time of 10 seconds, with a Minimum Speed of 10mm/s. It's not in the least bit unexpected that the time is nearly double what you're expecting if you've [...]

Thanks for the feedback.  Why then did it print 2 subsequent layers (of wall only) after initial + bottom at about 66 mm/s then drops below 40 mm/s for the remaining 346 layers as seen in the video?  As for the seam issue with the map,  I doesn't work for me.  Restarting the Cura shows the seam location specified is back left (which should be Washington).  When I sliced it, upon preview, it starts in southern California.

Asterchades, 2023-12-09:
> Why then did it print 2 subsequent layers (of wall only) after initial + bottom at about 66 mm/s then drops below 40 mm/s for the remaining 346 layers as seen in the video?

That appears to be the result of those two "layers" being treated as one - a solid, full-height loop at a fixed Z value gets laid down first followed by another that now has a slowly increasing Z throughout its duration before the "layer" is considered to be complete. The combined time taken to run the two loops is sufficiently long that it doesn't need to slow down as much, but it does still slow down to 52.73mm/s (F3163.5) before finishing the remainder at 25.47mm/s (F1528). _It's not exactly half the speed as the second loop in this faster layer appears to extend slightly beyond the seam it creates, making that layer slightly longer than twice the others._

Whether they should be treated as one is something I cannot answer to any meaningful degree of confidence, but they are - and that includes for the purposes of the Minimum Layer Time.

Unfortunately I can offer no further insight into your seam placement issue. I've been completely unable to replicate it - it updates correctly with the client open, and even relaunching the client doesn't cause it to revert without updating the setting.

tommyhp2, 2023-12-09:
Here's a screen shot of the wall seam.  Perhaps I'll file a separate bug report.
[image]

GregValiant, 2023-12-09:
Good morning to you both.
In regard to the cube and print speed...
I agree that it's all about the minimum layer time.
In regards to the quicker "first layer" of spiral - that occurs because "spiralize" doesn't start at the beginning of a layer, it starts mid-layer and as Asterchades says, there is one loop at a fixed Z and a second loop at a rising Z as the spiral actually starts.  Here is a snippet from the gcode at the transition.  This is within LAYER:2 in the gcode (layer 3 in the preview).
G1 X140.874 Y149.74 E871.87887
G1 X141.395 Y148.383 E871.93012
G1 X142.048 Y147.103 E871.98078
G1 X142.839 Y145.885 E872.03198
G1 X143.747 Y144.763 E872.08287
G1 X144.763 Y143.747 Z0.601 E872.13353
G1 X145.885 Y142.839 Z0.602 E872.18441
G1 X147.103 Y142.048 Z0.603 E872.23562
G1 X148.383 Y141.395 Z0.605 E872.28628
So that layer is twice as long (physically and time-wise) as the layers that follow and so it can print faster without violating the Minimum Layer Time.  Cura works with layers and so the constantly changing Z of spiralize can't be shown accurately in preview because it's a helix and doesn't exist neatly "all in one layer".

In regards to that map - that model is a mess.  It has a lot of errors but the glaring error is that it is poorly done.  Here we are in lovely southern Texas.  Look at the widths of the different portions of the "line".  The width of the model varies all along it's length from about 1mm wide down to about 0.17mm.  That is a very difficult slice so set up for.
[image]

This is the "as is" project file.  Lots of seams to go around and you can see there are gaps in the model where the model is thinner than the standard .34 minimum line width and simply couldn't slice.  I have a feeling that with all the starts and stops this would be a nightmare of strings.
[image]

Here it is with a 0.3 nozzle and in Surface mode.  There is the Z seam in southern California.
[image]

I don't see any bug here in either case.  The first model is all "expected behavior" given the settings.  Dropping the "Minimum Layer Time" to 1 second will give you a print at 66mm/sec.  My E3Pro with a 5051 blower wouldn't have the cooling for that.  There is a reason there is a Minimum Layer Time.

The second model is poorly designed and is a difficult piece to set up in Cura.  Once it is set up to slice completely it's all good (or at least as good as this will get).  The logical places for the Z seam are Maine and Washington as those are the two "endpoints" of the model but the Z seam can be anywhere so long as the rest of the Cura setup will allow this to slice without starts and stops.  There is no bug just because the results with a specific set of settings doesn't match your expectations.  A model like this requires careful thought in regards to how you want to go about setting the slice up to get a good result.

Some models just require their own brand of love.  For this "map" model, if I was to print it I'd probably go with a 0.6 nozzle at 0.8 line width and in "Surface Mode".  I wouldn't really care where the Z seam ended up (and my wife would never notice anything but the color anyway).

The bottom line is that I'm going to remove the bug label from this as I just don't see one and I'll close this as well.  If you feel that is premature let us know.  If you can explain where something is broken It can always be re-opened.

closed 2023-12-09 by GregValiant

tommyhp2, 2023-12-09:
Hi Greg,

Thanks for the feedback and detailed explanation.  I had to create the cubic model to isolate the causes (z wobble, over/under extrusion, x and/or y movements) z band issue that I've been trying to troubleshoot.  Because of that varying speed, it threw me off a for the same extrusion/flow.  I'll change both minimum time and speed as suggested.

As for the map, looking at the original picture online seems to show the line width was consistent enough that's why I chose it test how Cura would handle a single line which is similar to speed issue I've experience with my pseudo cube.  I guess all that conversion from webp to png to SVG (provided free by online service) probably messed up that line width badly.  This is also my first time converting a 2D image into a 3D model.  As for the seam location, I don't know why it always starts in southern California for me regardless of my attempts:

- change of seam location
- hiding preference
- restarting Cura
- switching models (other models work as expected but this map always starts in SoCal)

I'll look for a better SVG conversion tool before I try this map again.

Thank you both for your time.