Ultimaker/Cura #17571: car slice failure

suatbatu, 2023-12-09:
### Cura Version

5.6.0

### Operating System

W11

### Printer

ultimaker 3

### Describe your model

pre repaired in netfabb

### Add your .zip here ⬇️

error while slicing
[arabaaaa1 (repaired).zip](https://github.com/Ultimaker/Cura/files/13623557/arabaaaa1.repaired.zip)

GregValiant, 2023-12-09:
Thanks for the report.
This might be a problem between Cura and Windows 11 as I didn't have any problem slicing it in Win 10.  @5axes is keeping a list.
I'll mark this as a duplicate for the Cura team.

Here is the slice.  Even with "Remove all holes" turned on, it looks really bad and is not really printable.
[image]

GregValiant, 2026-10-01:
I'll close this.

closed 2026-10-01 by GregValiant