Ultimaker/Cura #17573: Slicing failure

yaturner, 2023-12-09:
### Cura Version

5.6.0

### Operating System

Ubuntu 22.04

### Printer

Creality Ender 3 Pro

### Name abnormal settings

N/A

### Describe model location

N/A

### Describe your model

No

### Add your .zip here ⬇️

[CE3PRO Santa String Art 0.24 Generic PLA lw 0.4mm lh 0.2mm if 20% ext1 210.0C bed 60C.zip](https://github.com/Ultimaker/Cura/files/13624385/CE3PRO.Santa.String.Art.0.24.Generic.PLA.lw.0.4mm.lh.0.2mm.if.20.ext1.210.0C.bed.60C.zip)

5axes, 2023-12-11:
Model issue : 
[image]

yaturner, 2023-12-11:
I can confirm that it slices correctly in Windows 11, so perhaps the problem is in the Linux version

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic