Ultimaker/Cura #17575: Failed slicing with tree support - success after changing model orientation - Tree supports

otede, 2023-12-09:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Custom FFF

### Name abnormal settings

The model slices with default DRAFT profile. Changes - compare yourself. Tree support is most probable culprit - the model slices with Regular.

### Describe model location

Yes, one location slices, the other doesn't. See the attached project files.

### Describe your model

Yes, one location slices, the other doesn't. See the attached project files.

### Add your .zip here ⬇️

High CPU usage around the middle of the process, then error after few seconds.
[LumePlus vs C5_6 slicing error.zip](https://github.com/Ultimaker/Cura/files/13624567/LumePlus.vs.C5_6.slicing.error.zip)
[image]
[image]
[image]

Prior to slicing the model has been scaled up proportionally by entering 120 mm for the Z size.

otede, 2023-12-09:
A non-standard printer
https://lumeplus.wixsite.com/lume

5axes, 2023-12-11:
An other Tree Support issue.. Duplicate https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

Slices, if the model is at the "right" place and we have enought distance to create the support.
[image]

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

otede, 2024-06-16:
I have re-opened the mech Delta_40mm_Autorifle_1 N04 projekt slicing error - DOESNT SLICE.3mf project and it indeed does slice. Well done, Cura Team!
[image]