Ultimaker/Cura #17577: Cura Slicing failed everytime

JadenM24, 2023-12-09:
### Cura Version

5.6.0

### Operating System

windows 10 pro

### Printer

kobra max 

### Name abnormal settings

no

### Describe model location

standing up 90 degrees 

### Describe your model

mask that i always print at 90 degrees 

### Add your .zip here ⬇️

Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker this always happens after you all release a new version of Cura, 5.5.0 It was great then 5.6.0 came out, and then I couldn't slice anything that I normally could so I installed 5.6.0 and still error.

5axes, 2023-12-12:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant