Ultimaker/Cura #17578: I just stumbled across this. Slices fine with regular supports. Tree supports crashed it.

davyjonezz, 2023-12-09:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

FL Sun FQ5

### Name abnormal settings

changing from normal supports to tree supports crashed it

### Describe model location

I didn't try. Just going to use normal supports

### Describe your model

It slices fine with normal supports

### Add your .zip here ⬇️

[FQ5_Short Rod - 10 sided split.zip](https://github.com/Ultimaker/Cura/files/13625563/FQ5_Short.Rod.-.10.sided.split.zip)

5axes, 2023-12-11:
Duplicate : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

Ok with the "right" tree support parameter. 
[image]

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0